carstenscharlemann/Climb-Obc-Hardwaretest #4: The OEM13085UL available in FHWN do not work as debugger with MCU 11.0.1

RobertK66, 2019-10-25:


RobertK66, 2019-10-26:
It seems somebody messed around with fimware update. It shows an MBED Drive with fail.txt containing  `SWD Error`

RobertK66, 2019-10-26:
not sure if all (4) boards have the same problem.

RobertK66, 2019-10-27:
Other than mine, the boards from WN have Hardware revision D2. So I upgraded the firmware to the same version than mine and it works ok. Upload of detailed descritpion / doku of board and firmware.bin file will follow.

WolfgangTreb, 2019-10-29:
I set the jumper and the explorer opens but without a firmware.bin file. Further, I have no access to copy files onto the board (I tried to copy the new firmware.bin file). Any idea what am I doing wrong?

WolfgangTreb, 2019-10-30:
I set the jumper next to the USB Interface. Without jumper it is the same. There are 5 smaller jumper occasions, which did not work out neither

RobertK66, 2019-10-30:
Ah now I see, The 'Jumper J4' here refers to 2 pins which have no 'jumper socket' soldered on. It is only written on the board. (If you hold it USB left side its on the left part of the PCB just right of the 12.000 MHz Quarz (Y4).  You have to contact the 2 pins per hand or solder some wires to it to get it connected.

RobertK66, 2019-10-30:
Here is a 'official' Document on firmware Updating: 
[LPCXpresso1769_CD_revD_ProductNote.pdf](https://github.com/carstenscharlemann/Climb-Obc-Hardwaretest/files/3790679/LPCXpresso1769_CD_revD_ProductNote.pdf)

WolfgangTreb, 2019-11-06:
Hi, It worked out. Just to mention (document): The jumper (J4) has to be replaced again before reconnecting the board and finding it with the MCU debugger

WolfgangTreb, 2019-11-06:
I will prepare the 3 debugger boards at FHWN in order to distribute them to students if applicable...

RobertK66, 2019-11-07:
For future references/doing:
You can close an issue with the left button close&comment once everything is resolved. (After closing an issue  all conversation are kept but filtered out from standard view of issue folder). If somebody. disagrees with your closing it can be reopened every time.

closed 2019-11-07 by RobertK66