carstenscharlemann/Climb-Obc-Hardwaretest #30: make RGB-LED work

RobertK66, 2019-12-07:
On the OBC we have a RGB LED WS2812 connected to PIN67 (IO Port P2[6]).
It would be nice to have a function available to control this LED by means of setting RGB value.

(on the LPCX board there is a RGB led with 3 output pins able to get 7 different colors. We also could write an additional  'Hardware abstraction' here for having same routines work on both boards...)

RobertK66, 2019-12-25:
@miguelcard : Have you had a chance to accire a WS2812 LED yet? If not we could arrange for a sessione where we test yourt routine(s) on the OBC I have..... (maybe remote is enough for that)

miguelcard, 2020-02-16:
@RobertK66 HI Robert this friday I will be at the university solving that and I will tell you then how it went, sorry for the late reply I have been a bit busy

RobertK66, 2020-03-12:
Hi @miguelcard,  another Friday needed 🤣 ?